daphtdazz/WebBLE #4: Can only write values to characteristic if writeWithoutResponse is enabled

cgreenhill, 2017-10-26:
Using writeValue on a BluetoothRemoteGATTCharacteristic always tries to use the writeWithoutResponse method, even if characteristic.properties.writeWithoutResponse is set to false, causing the write operation to fail.

Here's the error message received:
[CoreBluetooth] WARNING: Characteristic <CBCharacteristic: 0x1c80a8820, UUID = 2345, properties = 0xA, value = <f09f8c88>, notifying = NO> does not specify the "Write Without Response" property - ignoring response-less write

udox-cameron, 2018-03-20:
I am having the same problem. Write operation fails and have to use writeWithoutResponse to get it working.

daphtdazz, 2018-09-17:
Quick update (with apology for having taken so long), I have got a fix for this and should be getting it out in a week or two once I've had a chance to test it properly.

daphtdazz, 2018-09-25:
Just had version 1.1.6 approved on the app store which fixes this.

daphtdazz, 2018-09-25:
If anyone is able to test it for their device and let me know if it worked that would be marvellous.

daphtdazz, 2019-01-02:
Pretty sure that this is now fixed so am closing.

closed 2019-01-02 by daphtdazz