daphtdazz/WebBLE #5: [Feature Request]

rajsite, 2017-12-21:
It would be great to have Physical Web Eddystone beacon support enabled by the app as well.

Enabling users to discover and connect to devices around them broadcasting physical web urls and accepting ble connections. 🌈

daphtdazz, 2018-09-25:
@rajsite do you have any docs you can link me to that indicates what would be required here / any more detail about what extra stuff is needed? If it's a lot of work I'm unlikely to get to it any time soon unless a lot of people ask for it.

gfwilliams, 2018-09-26:
The main reference is here: https://github.com/google/eddystone/blob/master/protocol-specification.md

But basically you'd have to:

* Start scanning for Bluetooth LE advertising packets.
* Parse any that have the `0xFEAA` service data
* Look at the first byte of the `0xFEAA` service data, and if `(data[0]&0xF0)==0x10` then it's a URL
* Decode according to https://github.com/google/eddystone/tree/master/eddystone-url and then put a link to it somewhere in the UI

It'd be nice if this were something that could be done on the lock screen (I think Chrome on iOS manages it even if you have to explicitly enable it?) but then I imagine that probably requires a lot of fiddling.

While cool, it might actually be more interesting to implement https://webbluetoothcg.github.io/web-bluetooth/scanning.html (or a subset of it). Nobody actually seems to implement it at the moment though so I imagine it's a bit of a moving target! https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md#scanning-api

If you did that, you could just implement an Eddystone scanner in JS on Web BLE's homepage, then you don't have to worry about any of the iOS UI issues.

rajsite, 2023-07-15:
Think we can close the issue as the Google Nearby Notifications service ended in 2018 lol: https://killedbygoogle.com/#:~:text=Google%20Nearby%20Notifications

Looks like scanning also got permanently stuck `¯\_(ツ)_/¯`: https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md#scanning-api

closed 2023-07-15 by rajsite