daphtdazz/WebBLE #6: 'characteristicvaluechanged' callback doesn't seem to work

PofNClimb, 2018-03-20:
Once notifications are enabled, and a 'characteristicvaluechanged' callback has been set, it doesn't seem to be triggered. This issue has been noted with [this application](https://1oginov.github.io/Web-Bluetooth-Terminal/) that works perfectly with Google Chrome on Android and Windows 10 (with [this implementation](https://github.com/urish/web-bluetooth-polyfill)). The same issue is also noted with a personal code. Any idea of what's going on ?

jiboune, 2018-04-18:
+1 Same problem

louisn, 2018-07-24:
+1 Same problem.

louisn, 2018-07-24:
When setNotifyValue is executed on the Core Bluetooth Peripheral, it is expected that didUpdateNotificationStateFor will execute with success or error.  This never happens.  Also, didUpdateValue is never executed either.  This is at the core (pun intended) of the issue.  So far no solution for it that I've found.

daphtdazz, 2018-07-27:
Hi all, sorry I haven't been very proactive on this, I'm going to prioritise it in the next week. I will keep you updated.

daphtdazz, 2019-01-02:
Hi @louisn, I think I should finally find time this week to fix this. I had been concentrating on getting the console implemented. It would be really helpful if you could get a repro using the latest version of WebBLE and with console logging, and then send me the logs, as I'm not quite sure what the issue is yet.

louisn, 2019-01-02:
Upon further review, the previous comment isn't exactly the issue but still might be useful background information so I'm not deleting it.  Yet.

The picker acted a bit differently than I was expecting but it does allow me to select the Moxy Monitor once the Done button is pressed.

louisn, 2019-01-02:
Getting closer to some useful logs.  Here is the use case.  I don't see the js executions.  I don't see js execution for getCharacteristic values.  I've added the logging for more detail.  Something in how the transaction manager is setting up handlers perhaps?  I tried to fix it but don't understand that well enough.

> 2019-01-02 08:52:35.879113-0600 WebBLE[340:20658] triage for transaction Transaction(id: 279, key: device:getPrimaryService)
2019-01-02 08:52:35.879884-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 279);
"
2019-01-02 08:52:35.881335-0600 WebBLE[340:20658] triage for transaction Transaction(id: 281, key: device:getPrimaryService)
2019-01-02 08:52:35.882110-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 281);
"
2019-01-02 08:52:35.883531-0600 WebBLE[340:20658] triage for transaction Transaction(id: 283, key: device:getPrimaryService)
2019-01-02 08:52:35.884126-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 283);
"
2019-01-02 08:52:35.887491-0600 WebBLE[340:20658] triage for transaction Transaction(id: 285, key: device:getPrimaryService)
2019-01-02 08:52:35.888340-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 285);
"
2019-01-02 08:52:35.889676-0600 WebBLE[340:20658] triage for transaction Transaction(id: 287, key: device:getPrimaryService)
2019-01-02 08:52:35.890309-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 287);
"
2019-01-02 08:52:35.892646-0600 WebBLE[340:20658] triage for transaction Transaction(id: 289, key: device:getCharacteristic)
2019-01-02 08:52:35.893190-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 289);
"
2019-01-02 08:52:35.900012-0600 WebBLE[340:20658] triage for transaction Transaction(id: 291, key: device:getCharacteristic)
2019-01-02 08:52:35.900542-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 291);
"
2019-01-02 08:52:35.905284-0600 WebBLE[340:20658] triage for transaction Transaction(id: 293, key: device:getCharacteristic)
2019-01-02 08:52:35.905834-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 293);
"
2019-01-02 08:52:35.907337-0600 WebBLE[340:20658] triage for transaction Transaction(id: 295, key: device:getCharacteristic)
2019-01-02 08:52:35.907823-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 295);
"
2019-01-02 08:52:35.912960-0600 WebBLE[340:20658] triage for transaction Transaction(id: 297, key: device:getCharacteristic)
2019-01-02 08:52:35.913402-0600 WebBLE[340:20658] --> execute js: "window.receiveMessageResponse(true, "Success", 297);
"
2019-01-02 08:52:35.917190-0600 WebBLE[340:20658] triage for transaction Transaction(id: 299, key: device:readCharacteristicValue)
2019-01-02 08:52:35.918062-0600 WebBLE[340:20658] self.peripheral.readValue for transaction Transaction(id: 299, key: device:readCharacteristicValue)
2019-01-02 08:52:35.919913-0600 WebBLE[340:20658] triage for transaction Transaction(id: 301, key: device:readCharacteristicValue)
2019-01-02 08:52:35.920957-0600 WebBLE[340:20658] self.peripheral.readValue for transaction Transaction(id: 301, key: device:readCharacteristicValue)
2019-01-02 08:52:35.922335-0600 WebBLE[340:20658] triage for transaction Transaction(id: 303, key: device:readCharacteristicValue)
2019-01-02 08:52:35.923092-0600 WebBLE[340:20658] self.peripheral.readValue for transaction Transaction(id: 303, key: device:readCharacteristicValue)
2019-01-02 08:52:35.925575-0600 WebBLE[340:20658] triage for transaction Transaction(id: 305, key: device:writeCharacteristicValue)
2019-01-02 08:52:35.927727-0600 WebBLE[340:20658] triage for transaction Transaction(id: 307, key: device:readCharacteristicValue)
2019-01-02 08:52:35.928558-0600 WebBLE[340:20658] self.peripheral.readValue for transaction Transaction(id: 307, key: device:readCharacteristicValue)

daphtdazz, 2019-01-04:
Hmm, I'm struggling to reproduce the issue. I am able to `readValue()` ok and get responses. Are you also subscribing to value change notifications on the characteristic you're reading from? Or simply reading directly?  Will keep looking over the weekend and if you get anymore logs they'd be appreciated, especially if you are able to enable `nslog()` in WBUtils.js and send the javascript console output as well!

louisn, 2019-01-05:
Thanks for the js logging tip.  I'll add that and update the output.

louisn, 2019-01-14:
Still setting this up - I wanted to get the clearest data possible for posting.
One observation though.  When I use Safari Develop to connect to the iPad for browser troubleshooting something really strange happens.  The application I'm attempting to use polls the BLE device every few seconds to update display information.  Each poll opens a new Web Inspector window! 

The inspector window title is as follows
Web Inspector - Louis' IPad Mini - WebBLE - JSContext

Does that indicate that a new document or other context is opened each poll?  If so it could be that the BLE device or the transaction management is out of context?

I'll continue to attempt to get additional data from the logging output from a clean connection that demonstrates the issue.

daphtdazz, 2019-01-14:
That sounds weird! When you say "poll" do you just mean you are calling `readValue()` every few seconds? Or some combination of `writeValue()` and `readValue()`, or something else?

Also, another option for handling this might be for me to get one of the devices that you're using and give it a go myself with your code? If they're readily available and not too expensive I could buy one, or better yet if you have a spare you could send me one :-)

louisn, 2019-01-14:
I'll see what I can arrange with the company I'm working with.

louisn, 2019-01-15:
I can get a MoxyMonitor device to you.  Can you provide a shipping address to me at louis.naber@gmail.com.

daphtdazz, 2019-06-19:
Pretty sure this is fixed in 1.2.2 so closing.

closed 2019-06-19 by daphtdazz