daphtdazz/WebBLE #17: Premature disconnection with Samsung Gear VR controller

daphtdazz, 2019-04-02:
When trying to use a Samsung Gear VR controller with WebBLE the device disconnects after 2 or 3 seconds. On Chrome on the Mac it doesn't.

Perhaps there's some bit of the spec not implemented that is designed to handle this, or perhaps there's some part of the Core Bluetooth APIs that can be configured to avoid this. Needs some more research.

vmarchetti3, 2019-04-09:
I noticed your recent update and gave this another try today. Sadly, no dice. It's still disconnecting after a few seconds. It looks like you added quite a bit of support though! Thanks for all of your hard work!

daphtdazz, 2026-10-22:
It's been a while now, and without a better repro I don't think this can really go anywhere, so I'm going to close.