daphtdazz/WebBLE #19: Make geolocation work

daphtdazz, 2019-06-30:
Currently geolocation doesn't work, hopefully just need to add a privacy request for it so people can allow WebBLE to access the location services.

gfwilliams, 2019-07-01:
Is there a generic 'privacy request' API you can implement? There's other stuff (webcams, audio, etc) that would usually prompt the user in a similar way, and it'd be nice to be able to implement it all in one go (as well as enabling any new stuff that comes along later).

Gragamel, 2020-01-13:
That is not needed. The app needs to tell IOS/Android it uses location. If that is enabled stanrad java script calls just provide location. You hav similair for devices compass and sensors. Theu do not require a per app setting but just genertae a pop-up the first time you use them in a webpage.

daphtdazz, 2020-01-13:
Thanks @Gragamel , assuming it's that easy I can add it to the next release.

Gragamel, 2020-06-10:
Dear David,

Were you able to add the location feature to the app? I just finished the development of the WEBapp game  I made and now it is time to start communicating with my BT laser tag device for the final bits on IOS.

Regards,

daphtdazz, 2020-10-19:
Finally I have released 1.3.1 which supports the standard geolocation APIs:

https://developer.mozilla.org/en-US/docs/Web/API/Geolocation

closed 2020-10-19 by daphtdazz