daphtdazz/WebBLE #24: getDescriptor issue

petrhomolka, 2019-11-05:
Hello first thanks for this great app.
I Wrote application for reading serial data from my BLE device. According to debug console it seems there is some problem with connection but if I continue with reading I can see all my data in console without problem. Unfortunately these data are not forwarded to my application, I can see them only in console.
So my problem is that I didn't get the data to my application but they are readable only in console.

[image]
[image]
[image]

daphtdazz, 2019-12-11:
@petrhomolka Would you be able to send a small amount of your client code that reproduces the issue? And indicate where in the flow the logs appear?

petrhomolka, 2020-01-02:
Hi you can look to source code at: https://madesoft.de/meier/ That's the demo application where I'm testing it. It seems that error comes at line 44 (ble.js) where I'm trying to get descriptor. Second printscreen is after using readIMData function at line 66 (ble.js). Data was read correctly (in console are correct data in 3th printscreen) but wasn't forwarded to my application for parsing. (It is working well on Android so I guess that code itself is ok)

daphtdazz, 2020-02-16:
Ah, that's because `getDescriptor()` isn't implemented by WebBLE yet (as per the [readme](https://github.com/daphtdazz/WebBLE/blob/master/README.md)). I'll make this issue into a feature request. Not sure when I'll have time to get onto it though.

petrhomolka, 2020-02-27:
Hi thanks for the answer. I saw that it wasn't implemented yet. Strange for me is that I received correct data to debug console and only problem is that they are not propagated to application itself.