daphtdazz/WebBLE #27: can't find variable BluetoothRemoteGATTService

rajsite, 2019-12-05:
Only a few of the Web Bluetooth constructor functions are [exported in WebBLE](https://github.com/daphtdazz/WebBLE/blob/3a767b7ad382323cc95b1ef43cb16ba528071cf6/WBCore/Polyfill/WBPolyfill.js#L369) however my application attempts to use those constructor functions to perform `instanceof` checks for [validating parameters](https://github.com/rajsite/webvi-experiments/blob/32ca88c221f88fc0aab373ba2b06027a9f0d4633/WebBluetooth/WebBluetooth.gcomp/Support/WebBluetooth.js#L183) in some cases.

This results in the error:
```
Can't find variable: BluetoothRemoteGATTService
```
A link to the published page can be [found using githack](https://rawcdn.githack.com/rajsite/webvi-experiments/3746b561859afc63ffe743c365d1bfdcfdfa932b/WebBluetooth/PuckJSBlink.html).

daphtdazz, 2019-12-11:
Thanks, will fix in next release

daphtdazz, 2019-12-11:
For now (bit of hack) you should be able to access it (and others) at `uk.co.greenparksoftware.wb.BluetoothRemoteGATTService`

daphtdazz, 2020-02-18:
fixed in 1.3.0

closed 2020-02-18 by daphtdazz