daphtdazz/WebBLE #28: device disconnected: the connection has timed out unexpectedly

rajsite, 2019-12-05:
If I run the following page served from githack, click the `Connect to Puck.js` button, and then click `Toggle LED` you can see a connected Puck.js have the LED turn full white and then get the title error message:

```
Device disconnected: The connection has timed out unexpectedly
```

Looking at the [writeValue function in my application](https://github.com/rajsite/webvi-experiments/blob/441daff8cf6d902dfcb8997d0c2655c779d24f51/WebBluetooth/WebBluetooth.gcomp/Support/WebBluetooth.js#L208) I am aware that the value that is passed in is a Uint8Array TypedArray that is a view on a much larger buffer from a WebAssembly module's memory.

If I create a copy of the Uint8Array so that it has it's own much smaller buffer before passing it to WebBLE's implementation of writeValue then no error occurs and the buffer passes through correctly.

My guess is that passing a TypedArray view to the underlying WebAssembly modules memory is not behaving well in the WebBluetooth polyfill implementation.

daphtdazz, 2019-12-11:
Interesting, thanks for reporting it, will have a look when I get a chance.

daphtdazz, 2019-12-11:
By the way @rajsite, just to confirm, this works fine on e.g. Chrome on the desktop?

rajsite, 2019-12-12:
Yep yep! It does work without creating a new buffer on latest stable Chrome on Windows Desktop and Android.

daphtdazz, 2021-06-01:
This was fixed in 1.4.0. However, when using puck js (and probably other Nordic socs) you may find that sending any more than 20 bytes at a time causes a crash, which seems to be a bug in the firmware, which may be fixed in recent updates. But the issue of WebBLE using the underlying ArrayBuffer instead of the TypedArray is fixed (and 1.4.0 has been approved by Apple and is available now on the App Store).

closed 2021-06-01 by daphtdazz

rajsite, 2021-06-01:
@daphtdazz thanks! I appreciate the update!