daphtdazz/WebBLE #39: Support App Clip

DanTheMan827, 2021-04-11:
It would be really nice if there was a special URL that could be linked to that would open a web clip version of the app while navigating to the URL given

daphtdazz, 2021-08-08:
Sorry, I'm not really sure what you mean. If you're talking about adding web clips to the home screen as described [here](https://www.itouchapps.net/how-to-create-iphone-web-clips/), then I don't think that's going to be possible as it's not possible to specify a different browser to render those from Safari. 

What you can do though is use the `webble://` URI scheme to open a page in webble from Safari, so you could have a webpage with this in:

```html
<a href="webble://mydomain.com/webble/path/">Open in WebBLE!</a>
```

And then clicking on that would cause WebBLE to open at that URL. Might that help?

Will close this shortly unless I hear back.

DanTheMan827, 2021-08-08:
I’m referring to the app clip feature that Apple added.

For example, you could have a special URL that someone could share that goes to a page hosted on your domain, in the app you could have it detect that url and jump into the app… that’s all pretty standard stuff and works fine if they already have the app

But with an App Clip banner on the page you could let someone open the app clip version of the app even if they don’t have it installed, iOS would grab the app clip version and open it directly to the URL as if it were the full app responding to a url assigned to it

These special URLs when shared would pop up the app clip directly upon being tapped asking the user if they want to use it

Here’s some information https://developer.apple.com/app-clips/

daphtdazz, 2021-08-09:
Ah, thanks that makes sense. Not sure when this might be done but it's on the list now!