daphtdazz/WebBLE #46: Cant select a device on iOS app

na55imK, 2021-12-08:
Hi,

the  dialogue to select a BLE device, wont show anything, as shown in the picture. It is not possible to select anything. This happens on iPhone 8 iOS 15.1 and iPadOS 15.1.

[image]
 
I would be happy if there is a solution to this problem.

daphtdazz, 2022-03-27:
Hi @na55imK, did you manage to find a way around this? Were you able to connect from a different browser, and what did that look like? What device were you trying to connect to?

na55imK, 2022-09-26:
Hi, sorry it turned out it was due to a defect Bangle.JS 2.
It works like charm now. Thank you for your response!

closed 2022-09-26 by daphtdazz