daphtdazz/WebBLE #48: reconnect support

aspektxxx, 2021-12-17:
It seems that gatt.connect() call is not possible after gatt.disconnect(). I'm getting following error: 
No known device for device transaction Transaction(id: 33, key: device:connectGATT), 
where the 33 is some internal number of the request.

The same code works on both chrome and edge on Windows.

Is this a bug or a missing feature?

daphtdazz, 2022-02-11:
Sounds like a bug!

murkle, 2023-07-21:
Here's the Google recommended way to handle auto-reconnect (seems to work well in Chrome without any additional flags 🤞)

https://docs.google.com/document/d/1RF4D-60cQJWR1LoQeLBxxigrxJwYS8nLOE0qWmBF1eo/edit

```
async function getPermittedBluetoothDevices() {
  let devices = await navigator.bluetooth.getDevices();
  for (let device of devices) {
    // Start a scan for each device before connecting to check that they're in
    // range.
    let abortController = new AbortController();
    await device.watchAdvertisements({signal: abortController.signal});
    device.addEventListener('advertisementreceived', async (evt) => {
      // Stop the scan to conserve power on mobile devices.
      abortController.abort();

      // Advertisement data can be read from |evt|.
      let deviceName = evt.name;
      let uuids = evt.uuids;
      let appearance = evt.appearance;
      let pathloss = evt.txPower - evt.rssi;
      let manufacturerData = evt.manufacturerData;
      let serviceData = evt.serviceData;

      // At this point, we know that the device is in range, and we can attempt
      // to connect to it.
      await evt.device.gatt.connect();
    });
  }
}
```

daphtdazz, 2026-11-09:
Fixed in the now-released 1.8.0

closed 2026-11-09 by daphtdazz