daphtdazz/WebBLE #49: Connecting to a device that's connected to the OS already

gfwilliams, 2022-02-17:
With Bangle.js you can now enable ANCS and AMS, which allows it to grab notifications and music playback info direct from iOS without an app installed (iOS stays connected to the Bangle). You may also have a HID bluetooth device connected to iOS that you also want to control with WebBLE.

However, when iOS is connected to the device, you don't see it listed in WebBLE and can't connect to it.

However on Android/other OSes you generally can do this.

I believe this should be possible on iOS too, and the BlueFruit app seems able to do it (if that's any help to see how they manage it?) https://github.com/adafruit/Bluefruit_LE_Connect_v2

daphtdazz, 2022-03-27:
@gfwilliams is there a bit of the web bluetooth spec that covers this do you know? How would you access these connected devices through the web APIs? Would you just expect to see them in `requestDevices()`?

gfwilliams, 2022-03-28:
Yes, they appear through `requestDevices` just like any other device. It's possible you may not be able to get hold of full advertising data, but I guess at minimum you'll know the device name.

In https://webbluetoothcg.github.io/web-bluetooth/#scan-for-devices it does mention:

> Let nearbyDevices be a set of [Bluetooth device](https://webbluetoothcg.github.io/web-bluetooth/#bluetooth-device)s, initially equal to the set of devices that are connected (have an [ATT Bearer](https://webbluetoothcg.github.io/web-bluetooth/#att-bearer)) to the UA.

Then it talks about adding devices that it's found by doing a scan...