daphtdazz/WebBLE #50: characteristic.writeValue not working

ankitnetsol, 2022-03-22:
Hello,

   We are using this browser to test BLE in IOS devices but when characteristic.writeValue comes into a role at that time it is stuck on this statement and does not work.

Anyone, Can you give me a solution.

Thanks.

daphtdazz, 2022-03-27:
Hi, could you provide some example code and what you are expecting to happen and what does happen?

closed 2022-04-04 by ankitnetsol

r03ert0, 2022-07-23:
I can provide an example:

I'm sending writeValue(2000), but the BLE device receives 208 (which is 2000 & 0xff). Using the same code in Chrome from a computer, or in Chrome from an android phone, the value received is 2000.

Here's the code:

```js
    const data = new Int32Array([2000]);
    myCharacteristic.writeValue(data)
```