daphtdazz/WebBLE #61: Unable to use self-signed certificates.

sigmaxipi, 2024-11-19:
I'm doing some local development and setup a webserver with a self-signed cert. I got it loaded into iOS's keystore and can access it via Safari. However, I get **An SSL error has occurred and a secure connection to the server cannot be made.** error in the app. I tried with both Deno's HTTPS server and your `https.py` script and both have the same issue where I can access them in Safari on the phone, but not WebBLE. I was able to get it working via a change to Info.plist so I'm not blocked. But is this a known issue?

I'm using iOS 18.1 on an iPhone 12 Pro with the most recent App Store build.

daphtdazz, 2026-10-22:
I have been able to get self-signed certificates to work following the instructions in [https.py](https://github.com/daphtdazz/WebBLE/blob/master/https.py) (which mostly just points you at [Apple's helpful guide](https://developer.apple.com/library/archive/technotes/tn2326/_index.html)). It's a complicated area and since I have it working locally unless someone can show this is a problem with WebBLE rather than it just being difficult to get sorted out regardless of WebBLE I'm going to close this.

sigmaxipi, 2026-10-23:
This problem stopped happening about a year ago for me. Now I'm using the stock app from the store without any issue.

closed 2026-10-23 by sigmaxipi