daphtdazz/WebBLE #62: Cannot edit PyBricks block textboxes with WebBLE on IPad

ralrivera, 2024-12-04:
**Describe the bug**
WebBLE is the only browser so-far that I have been able to connect with a Pybricks hub with on an IPad. Creating and deploying code works great! The only problem encountered so far is that you cannot edit text values of textboxes in blocks. We haven't tested every block, but it happens with set-up blocks names, motor power values, comparison values, & variable names/values. This is not the case with other IPad browsers. Chrome & Bluefy both can edit block textboxes when tested on the same IPad. The issue persists after restarting the program and restarting the IPad.

This happens both when using just the touch screen and when an external keyboard/trackpad is used.

**To reproduce**
Steps to reproduce the behavior:

Go to code.pybricks.com or beta.pybricks.com with the WebBLE app on an IPad
Create a new blocks project
click on the default "Hello, Pybricks!" textbox within the print block to edit it.
No keyboard or input method is displayed to edit.
click setup to add a hub block
drag and connect to setup stack
try to edit the Hub's name and get similar results to steps 3-4 above
Expected behavior
An input window should appear where you can type the desired value using a physical keyboard or a virtual keyboard should appear to input a value if no keyboard is present

**Screenshots**
None of the white 'greyed out' name boxes in setup or any of the white textboxes were able to be edited. All of the drop-down controls worked as expected.

[image]

**Extra Info**
WebBLE Version - 1.7.0
IPadOS Version - 17.5.1
Also created a Pybricks report: https://github.com/pybricks/support/issues/1963

laurensvalk, 2024-12-10:
Does it show any errors in the console?

Do you see similar issues in the upstream Blockly version, separate from Pybricks? You can try it here: https://blockly-demo.appspot.com/static/tests/playgrounds/advanced_playground.html

ralrivera, 2024-12-10:
Is the console the output on the bottom you see after hitting the button with the left justified lines? It says "creating code completion worker" & "code completion engine is ready". No other message pops up as I try to edit the text boxes.

The same issue occurs at the blocky link provided. All of the 'editable' boxes in the attached screenshot were not accessible in WebBLE but worked when tried on other browsers on the same IPad. The 'undefined is not an object' error seemed to be present in other browsers as well. 

One additional issue with the blockly site was that more than one variable could not be created. When you hit the new variable button in other browsers, a dialog box would open for you to name the variable. With WebBLE, no dialog box shows up to name it. Eventually a variable named 'x' appeared, I assume that is the default name after timing out waiting for a name. No other variable was able to be made through WebBLE since that dialog box does not appear.

ralrivera, 2024-12-10:
[image]

ralrivera, 2025-03-28:
Bumping to see if @daphtdazz has any thoughts. I'm currently using Pybricks in a 8th grade class and letting them edit on in-class computers. It would be convenient if they could use their provided Ipads to code outside of class. Wanted to check if there was potential support for this issue to determine how to proceed for future classes.

Thanks!

triplef, 2025-04-19:
+1 would be great to see this fixed to be able to fully use Pybricks on iPad. 🙏

daphtdazz, 2025-04-22:
Apologies for radio silence on this. I will try and have a look this week at this and the other pybricks issues.

daphtdazz, 2026-09-15:
Right, so I didn't manage to do that. however I'm here to say that I have finally got some time available in my life to look at these issues and I should be able to do so this autumn. I would very much like to be able to ensure that WebBLE works well for pybricks. As some evidence that I might be able to this time I have just pushed a bunch of minorish fixes and support for iOS 27 in version 1.7.1, so I have a bit of momentum...

daphtdazz, 2026-09-16:
Issue found, and I have a fix. Probably will aim to do another release next week.

daphtdazz, 2026-10-01:
This is now fixed in 1.7.1

closed 2026-10-01 by daphtdazz

mayerwin, 2026-10-01:
Fantastic, is there any way we can test it actually works well before purchasing?

ralrivera, 2026-10-02:
Thank you @daphtdazz! @mayerwin, I just checked on my device and it was working.

daphtdazz, 2026-10-08:
> Fantastic, is there any way we can test it actually works well before purchasing?

Well you could build the app yourself locally if you like! Otherwise no.

One thing I would say: downloads still don't work but I'm actively working on that so hopefully I'll have a release by the end of October done to support those (1.8 probably). Also I think due to the web framework that pybricks uses some things don't work very well on iOS, like some of the controls for file management only appear on mouse hover, which means they never appear on iOS as far as I can tell! It's on my list to see if there's anything I can do about that on my end.

daphtdazz, 2026-10-08:
Also @mayerwin FWIW I think it's relatively easy to get a refund for an app that doesn't work see [here](https://support.apple.com/en-gb/118223)